PJLab-ADG/SensorsCalibration #129: What is the right Lidar-camera calibration approach specific to the requirements?

muditshrivastava, 2023-11-17:
We are using Ouster OS1 Lidar data and four cameras data for sensor fusion purpose. We want to calibrate Lidar-camera to obtain output transformation matrix. Please suggest the right Lidar camera calibration approach among auto-calib using segment anything, Joint calib and manual calib.

Why we are not getting any output transformation matrix using joint calib approach. Also PCL 1.9 exact version is not able to be installed, instead the latest version of PCL got installed that is not supported by the algorithm.

Lpy15540250240, 2023-11-23:
Just use manual calib model , i try it by ouster fusion，result will be fine.

muditshrivastava, 2023-11-23:
By using manual calib, the input matrix is fed into the algorithm which matches exactly with the output Lidar- camera transformation matrix.

Then, what is the use of getting output matrix, if we are already feeding the same  input matrix?