PJLab-ADG/SensorsCalibration #130: 在用cmake编译时，在make步骤后，出现这样的问题，大致好像是关于静态库libsplinter-static-3-0.a，请问有什么解决方法吗？十分感谢，我的环境是ubuntu18，Intel® Core™ i7-8700 CPU @ 3.20GHz × 12

rafion1, 2023-11-18:
/usr/bin/ld: /home/wx/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bspline.cpp.o): relocation R_X86_64_32S against symbol `_ZTVN8SPLINTER10SerializerE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wx/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bsplinebasis.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wx/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bsplinebasis1d.cpp.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wx/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bsplinebuilder.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wx/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(datatable.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wx/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(function.cpp.o): relocation R_X86_64_32S against symbol `_ZTVN8SPLINTER8SaveableE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wx/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(mykroneckerproduct.cpp.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wx/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(serializer.cpp.o): relocation R_X86_64_32S against symbol `_ZTVN8SPLINTER10SerializerE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wx/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(utilities.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wx/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(datapoint.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: 最后的链结失败: 输出不可表示的节

rafion1, 2023-11-18:
原来是我的电脑环境没有splinter，先编译splinter后，已解决

closed 2023-11-18 by rafion1

yesheng1996, 2024-03-19:
请问你的splinter是如何安装的？

pedestrain123, 2024-10-10:
@rafion1 你好，我进入3rdparty编译安装了splinter，但是依旧编译报这个错误，请问怎么解决？

pedestrain123, 2024-10-10:
@yesheng1996 请问这个问题你解决了吗？