PJLab-ADG/SensorsCalibration #136: cc1plus: fatal error: CMakeFiles/run_lidar2camera.dir/src/run_lidar2camera.cpp.d: No such file or directory

963070827, 2024-01-07:
cc1plus: fatal error: CMakeFiles/run_lidar2camera.dir/src/run_lidar2camera.cpp.d: No such file or directory
compilation terminated.
CMakeFiles/run_lidar2camera.dir/build.make:75: recipe for target 'CMakeFiles/run_lidar2camera.dir/src/run_lidar2camera.cpp.o' failed
make[2]: *** [CMakeFiles/run_lidar2camera.dir/src/run_lidar2camera.cpp.o] Error 1
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/run_lidar2camera.dir/all' failed
make[1]: *** [CMakeFiles/run_lidar2camera.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

cumtbjt, 2024-01-29:
使用的docker环境 相同的报错

cumtbjt, 2024-01-29:
set(CMAKE_CXX_FLAGS "-g3 -O -Wall")
把CmakeLists的第三行的小o改成大O即可解决问题 成功编译
希望能够帮到你！