PJLab-ADG/SensorsCalibration #139: Compilation KO on debian

fghoussen, 2024-01-25:
As it seems there is no debian package for Pangolin, I build it from source:

```
/tmp/Pangolin/build> cmake -DCMAKE_INSTALL_PREFIX=/tmp/Pangolin/local
/tmp/Pangolin/build> make all install

/tmp/Pangolin> cat local/lib/cmake/Pangolin/PangolinConfig.cmake 
# Compute paths
get_filename_component( PROJECT_CMAKE_DIR "${CMAKE_CURRENT_LIST_FILE}" PATH )
# Library dependencies (contains definitions for IMPORTED targets)
if( NOT Pangolin_BINARY_DIR )
  include( "${PROJECT_CMAKE_DIR}/PangolinTargets.cmake" )
endif()
SET( Pangolin_CMAKEMODULES /tmp/Pangolin/cmake )
SET( Pangolin_LIBRARIES    pango_core;pango_display;pango_geometry;pango_glgeometry;pango_image;pango_opengl;pango_packetstream;pango_plot;pango_python;pango_scene;pango_tools;pango_vars;pango_video;pango_windowing;tinyobj )
SET( Pangolin_LIBRARY      "${Pangolin_LIBRARIES}" )
include(CMakeFindDependencyMacro)
find_dependency(Eigen3)
if (UNIX)
  find_dependency(Threads)
endif()

/tmp/Pangolin> git log -1
commit 099f298d785b757c8691b88a90210d99c5fcf143 (HEAD -> master, origin/master, origin/HEAD)

```

Now Trying to build this repo doesn't work:

```
/tmp/SensorsCalibration/lidar2lidar/manual_calib/build> git log -1
commit 1b701c11eed6c96e23fe13b98430a510a91d3498 (HEAD -> master, origin/master, origin/HEAD)

/tmp/SensorsCalibration/lidar2lidar/manual_calib/build> export Pangolin_DIR="/tmp/Pangolin/local/lib/cmake/Pangolin"

/tmp/SensorsCalibration/lidar2lidar/manual_calib/build> cmake ..; make
...
-- Configuring done (2.6s)
-- Generating done (0.0s)
-- Build files have been written to: /tmp/SensorsCalibration/lidar2lidar/manual_calib/build
[ 25%] Building CXX object CMakeFiles/LidarToLidar.dir/src/run_lidar2lidar.cpp.o
/tmp/SensorsCalibration/lidar2lidar/manual_calib/src/run_lidar2lidar.cpp:9:10: fatal error: pangolin/pangolin.h: No such file or directory
    9 | #include <pangolin/pangolin.h>
      |          ^~~~~~~~~~~~~~~~~~~~~

```

How to fix this compilation error?

fghoussen, 2024-01-25:
I have just tried to force include paths but compilation fails:

```
/tmp/SensorsCalibration/lidar2lidar/manual_calib/build> git diff
diff --git a/lidar2lidar/manual_calib/CMakeLists.txt b/lidar2lidar/manual_calib/CMakeLists.txt
index 9a191c9..0bcfc2a 100755
--- a/lidar2lidar/manual_calib/CMakeLists.txt
+++ b/lidar2lidar/manual_calib/CMakeLists.txt
@@ -1,9 +1,10 @@
 cmake_minimum_required(VERSION 2.8.3)
 project(LidarToLidar)
-set(CMAKE_CXX_FLAGS "-std=c++11 -g -Wall")
+set(CMAKE_CXX_FLAGS "-std=c++11 -g -Wall -I/tmp/Pangolin/local/include -I/usr/include/jsoncpp")
 set(CMAKE_CXX_STANDARD 14)

/tmp/SensorsCalibration/lidar2lidar/manual_calib/build> make
...
[ 25%] Building CXX object CMakeFiles/LidarToLidar.dir/src/run_lidar2lidar.cpp.o
In file included from /tmp/Pangolin/local/include/pangolin/var/var.h:37,
                 from /tmp/Pangolin/local/include/pangolin/display/widgets.h:31,
                 from /tmp/Pangolin/local/include/pangolin/pangolin.h:40,
                 from /tmp/SensorsCalibration/lidar2lidar/manual_calib/src/run_lidar2lidar.cpp:9:
/tmp/Pangolin/local/include/pangolin/var/varstate.h: In member function ‘std::map<std::__cxx11::basic_string<char>, std::shared_ptr<pangolin::VarValueGeneric> >::iterator pangolin::VarState::AddVar(const std::shared_ptr<pangolin::VarValue<T> >&, bool)’:
/tmp/Pangolin/local/include/pangolin/var/varstate.h:242:16: warning: structured bindings only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Wc++17-extensions]
  242 |     const auto [it, success] = vars.insert(VarStoreMap::value_type(var->Meta().full_name, var));
      |                ^
In file included from /tmp/Pangolin/local/include/pangolin/gl/gl.h:324,
                 from /tmp/Pangolin/local/include/pangolin/handler/handler.h:31,
                 from /tmp/Pangolin/local/include/pangolin/display/widgets.h:32:
/tmp/Pangolin/local/include/pangolin/gl/gl.hpp: In member function ‘void pangolin::GlTexture::CopyFrom(const pangolin::GlTexture&)’:
/tmp/Pangolin/local/include/pangolin/gl/gl.hpp:348:5: error: ‘glCopyImageSubDataNV’ was not declared in this scope
  348 |     glCopyImageSubDataNV(tex.tid, GL_TEXTURE_2D, 0, 0, 0, 0,
      |     ^~~~~~~~~~~~~~~~~~~~

```

lsangreg, 2024-03-13:
did you solve this?