PJLab-ADG/SensorsCalibration #141: 编译雷达2相机失败

cumtbjt, 2024-01-28:
/usr/local/include/pangolin/gl/glsl.hpp: In member function ‘GLint pangolin::GlSlProgram::GetProgramResourceIndex(const string&)’:
/usr/local/include/pangolin/gl/glsl.hpp:539:12: error: ‘glGetProgramResourceIndex’ was not declared in this scope; did you mean ‘GetProgramResourceIndex’?
  539 |     return glGetProgramResourceIndex(prog, GL_SHADER_STORAGE_BLOCK, name.c_str());
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~
      |            GetProgramResourceIndex
/usr/local/include/pangolin/gl/glsl.hpp: In member function ‘void pangolin::GlSlProgram::SetShaderStorageBlock(const string&, const int&)’:
/usr/local/include/pangolin/gl/glsl.hpp:544:5: error: ‘glShaderStorageBlockBinding’ was not declared in this scope
  544 |     glShaderStorageBlockBinding(prog, GetProgramResourceIndex(name), bindingIndex);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/lzl/SensorsCalibration/radar2camera/manual_calib/src/run_radar2camera.cpp:26:
/home/lzl/SensorsCalibration/radar2camera/manual_calib/include/projector_radar.hpp: In member function ‘void Projector::ProjectToRawMat(cv::Mat, cv::Mat, cv::Mat, cv::Mat, cv::Mat, cv::Mat&, cv::Mat&)’:
/home/lzl/SensorsCalibration/radar2camera/manual_calib/include/projector_radar.hpp:192:13: warning: unused variable ‘y’ [-Wunused-variable]
  192 |       float y = transCloud2d.at<float>(1, i);
      |             ^
/home/lzl/SensorsCalibration/radar2camera/manual_calib/include/projector_radar.hpp:218:13: warning: unused variable ‘d’ [-Wunused-variable]
  218 |       float d = points[i].dist;
      |             ^
make[2]: *** [CMakeFiles/RadarToCamera.dir/build.make:63：CMakeFiles/RadarToCamera.dir/src/run_radar2camera.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:105：CMakeFiles/RadarToCamera.dir/all] 错误 2
make: *** [Makefile:84：all] 错误 2
 请问该如何解决！

cumtbjt, 2024-01-28:
发现是pangolin包版本不兼容的问题 这里使用0.6版本即可编译！

L-Net-1992, 2024-03-18:
ubuntu22.04么？