PJLab-ADG/SensorsCalibration #142: radar2camera无法进入标定盘 一直卡在选择四个点 按enter键无效

cumtbjt, 2024-01-28:
运行示例之后一直卡在选择四个点界面 按enter无效 该如何进入下面的标定界面呢？

cumtbjt, 2024-01-28:
不管是用docker拉取镜像还是用自己的环境 都会开在选择四个点 没法进入下一阶段，如何进行标定 希望能够得到回答！感谢！！！！！！！！！！！！！

L-Net-1992, 2024-03-20:
> 不管是用docker拉取镜像还是用自己的环境 都会开在选择四个点 没法进入下一阶段，如何进行标定 希望能够得到回答！感谢！！！！！！！！！！！！！

修改这里：
void PickPoint() {
    cv::namedWindow("image", 0);
    cv::setMouseCallback("image", mouse_event, this);
    cv::imshow("image", image_);
    while (true) {
      int key = cv::waitKey(10);
      key = key % 255;

      // evaluate
      if (key == 13) // enter
      // if (key == 10) // enter
      {
        bool flag = computeHomo();
        if (flag)
          break;
      }
    }
    cv::destroyWindow("image");
  }

wycrystal, 2024-04-19:
> 修改这里： void PickPoint() { cv::namedWindow("image", 0); cv::setMouseCallback("image", mouse_event, this); cv::imshow("image", image_); while (true) { int key = cv::waitKey(10); key = key % 255;
>
> ``` [...]

请问这个是在哪个文件中修改的？

CHUDADA19, 2024-07-29:
@cumtbjt 解决了吗

1029626013, 2025-03-31:
可以加一个std::cout做测试：
void PickPoint() {
cv::namedWindow("image", 0);
cv::setMouseCallback("image", mouse_event, this);
cv::imshow("image", image_);
while (true) {
int key = cv::waitKey(10);
key = key % 255;
if(key != -1){
std::cout<<key<<endl;  //可以知道按键输出的值，更好修改，我测试得到enter值为10
}
  // evaluate
  //if (key == 13) // enter
  if (key == 10) // enter
  {
    bool flag = computeHomo();
    if (flag)
      break;
  }
}
cv::destroyWindow("image");
}