PJLab-ADG/SensorsCalibration #147: Error running the camera intrinsic calibration program run_intrinsic_calibration

Ahqiu1, 2024-03-12:
When I run the command.bin/run_intrinsic_calibration data/, an error occurs terminate called after throwing an instance of 'std::out_of_range' what(): basic_string::substr: __pos(which is 140) &gt;  this-&gt; size() (which is 0)

Tkk0124, 2025-01-03:
#include <iostream>
#include <dirent.h>

#include "IntrinsicCalibration.hpp"
#include <opencv2/opencv.hpp>

const char usage[] = 
    "\t./bin/run_intrinsic_calibration <calibration_image_dir> \n"
    "example:\n\t"
    "./bin/run_intrinsic_calibration ./data/\n";

int main(int argc, char** argv){
    if (argc < 2){
        std::cout << argv[0] << usage;
        return 1;
    }
    
    std::string calibration_image_dir = argv[1];
    if (calibration_image_dir.back() != '/'){
        calibration_image_dir += "/";
    }
    
    IntrinsicCalibration calibrator;
    calibrator.Calibrate(calibration_image_dir); 
    return 0;
}

Tkk0124, 2025-01-03:
// cv::Mat image = cv::imread(input_image_path.c_str(), 0);

Tkk0124, 2025-01-03:
Comment it out “cv::Mat image = cv::imread(input_image_path.c_str(), 0);”