PJLab-ADG/SensorsCalibration #155: lidar2car标定livox mid360点云数据，yaw角度无法标定出来

yueyihua, 2024-05-11:
--------------------------------------------------
start calibrating lidar yaw
No valid data for calibrating yaw.
Calibration result was saved to file calib_result.txt

yesheng1996, 2024-05-18:
请问你现在跑成功了吗？我现在是卡在
BSpline bspline_x = BSpline::Builder(sample_x).degree(bspine_degree_).smoothing(BSpline::Smoothing::PSPLINE).alpha(0.03).build();
    BSpline bspline_y = BSpline::Builder(sample_y).degree(bspine_degree_).smoothing(BSpline::Smoothing::PSPLINE).alpha(0.03).build();
每次到这里都是Segmentation fault (core dumped).

yueyihua, 2024-05-22:
> 请问你现在跑成功了吗？我现在是卡在 BSpline bspline_x = BSpline::Builder(sample_x).degree(bspine_degree_).smoothing(BSpline::Smoothing::PSPLINE).alpha(0.03).build(); BSpline bspline_y = [...]

1. Change map_resolution to 0.25 in fastLoam.cpp
2. My problem: data all continued in function YawCalib::GetYawSegs:
        if (dx * dx + dy * dy < 1e-5) {
            continue;
        }

samz00, 2024-06-26:
我也有一样的问题，似乎LOAM算出来的pose是0：
<-------------------Start running Lidar SLAM566 --------------------->edge:3380
surf:81766
callbacks.cc:125 iter      cost      cost_change  |gradient|   |step|    tr_ratio  tr_radius  ls_iter  iter_time  total_time
   0  9.252431e+00    0.00e+00    5.53e-01   0.00e+00   0.00e+00  1.00e+04        0    1.69e-04    2.13e-04
trust_region_minimizer.cc:428 Invalid step: current_cost: 9.25243 absolute model cost change: -535575 relative model cost change: -57884.7
callbacks.cc:125    1  9.252431e+00    0.00e+00    5.53e-01   0.00e+00   0.00e+00  5.00e+03        1    7.49e-05    3.12e-04
trust_region_minimizer.cc:428 Invalid step: current_cost: 9.25243 absolute model cost change: -535716 relative model cost change: -57900
callbacks.cc:125    2  9.252431e+00    0.00e+00    5.53e-01   0.00e+00   0.00e+00  1.25e+03        1    2.60e-05    3.41e-04
trust_region_minimizer.cc:428 Invalid step: current_cost: 9.25243 absolute model cost change: -536562 relative model cost change: -57991.5
callbacks.cc:125    3  9.252431e+00    0.00e+00    5.53e-01   0.00e+00   0.00e+00  1.56e+02        1    2.50e-05    3.68e-04
trust_region_minimizer.cc:428 Invalid step: current_cost: 9.25243 absolute model cost change: -544431 relative model cost change: -58841.9
callbacks.cc:125    4  9.252431e+00    0.00e+00    5.53e-01   0.00e+00   0.00e+00  9.77e+00        1    2.69e-05    3.98e-04
trust_region_minimizer.cc:636 Terminating: Maximum number of iterations reached. Number of iterations: 4.
callbacks.cc:125 iter      cost      cost_change  |gradient|   |step|    tr_ratio  tr_radius  ls_iter  iter_time  total_time
   0  9.252431e+00    0.00e+00    5.53e-01   0.00e+00   0.00e+00  1.00e+04        0    1.61e-04    1.96e-04
trust_region_minimizer.cc:428 Invalid step: current_cost: 9.25243 absolute model cost change: -535575 relative model cost change: -57884.7
callbacks.cc:125    1  9.252431e+00    0.00e+00    5.53e-01   0.00e+00   0.00e+00  5.00e+03        1    3.79e-05    2.44e-04
trust_region_minimizer.cc:428 Invalid step: current_cost: 9.25243 absolute model cost change: -535716 relative model cost change: -57900
callbacks.cc:125    2  9.252431e+00    0.00e+00    5.53e-01   0.00e+00   0.00e+00  1.25e+03        1    2.69e-05    2.74e-04
trust_region_minimizer.cc:428 Invalid step: current_cost: 9.25243 absolute model cost change: -536562 relative model cost change: -57991.5
callbacks.cc:125    3  9.252431e+00    0.00e+00    5.53e-01   0.00e+00   0.00e+00  1.56e+02        1    2.60e-05    3.02e-04
trust_region_minimizer.cc:428 Invalid step: current_cost: 9.25243 absolute model cost change: -544431 relative model cost change: -58841.9
callbacks.cc:125    4  9.252431e+00    0.00e+00    5.53e-01   0.00e+00   0.00e+00  9.77e+00        1    2.69e-05    3.31e-04
trust_region_minimizer.cc:636 Terminating: Maximum number of iterations reached. Number of iterations: 4.
这里trust_region_minimizer的absolute model cost 是一个异常小的值，不知道是什么原因

pedestrain123, 2024-10-10:
@yueyihua 你好请问您有用过这个标定工具箱里面lidar2ins自动标定算法吗？就是基于balm的那个方法

zhangyuran-gg, 2024-10-11:
> > 我现在是卡在 BSpline bspline_x = BSpline：：Builder（sample_x）.degree（bspine_degree_）.smoothing（BSpline：：Smoothing：:P SPLINE）.alpha（0.03）.build（）;BSpline bspline_y = BSpline：：Builder（sample_y）.degree（bspine_degree_）.smoothing（BSpline：：Smoothing：:P SPLINE）.alpha（0.03）.build（）;每次到这里都是 Segmentation fault （core dumped）。
> 
> 1. 将 map_resolution 更改为 0.25 in fastLoam.cpp
> 2. 我的问题：数据都在函数 YawCalib：：GetYawSegs 中继续： if （dx * dx + dy * dy < 1e-5） { continue;
>    }

Have you solved your problem？

pedestrain123, 2024-10-13:
> Have you solved your problem？

加个QQ交流一下吧？2985782675

kun110, 2025-03-26:
> 我也有一样的问题，似乎LOAM算出来的pose是0： <-------------------Start running Lidar SLAM566 --------------------->edge:3380 surf:81766 callbacks.cc:125 iter cost cost_change |gradient| |step| tr_ratio tr_radius [...]

麻烦问下，您这个解决了吗