PJLab-ADG/SensorsCalibration #157: Trouble building L2C auto_calib_v2

SST-bplus, 2024-06-05:
I am trying to use Lidar2Camera auto_calib_v2, but sadly I haven't come too far. 

After starting the docker container I  went through 

git clone https://github.com/OpenCalib/CalibAnything.git
cd CalibAnything
# mkdir build
mkdir -p build && cd build
# build
cmake .. && make

but it would not build. The console errors is too long to paste here, but it ends with

                      ^
/share/CalibAnything/src/calibration.cpp:601:31: warning: range-based 'for' loops only available with -std=c++11 or -std=gnu++11
     for (const auto &src_pt : pcs_[index]->points)
                               ^
/share/CalibAnything/src/calibration.cpp:604:23: error: request for member 'x' in 'src_pt', which is of non-class type 'const int'
         vec << src_pt.x, src_pt.y, src_pt.z, 1;
                       ^
/share/CalibAnything/src/calibration.cpp:604:33: error: request for member 'y' in 'src_pt', which is of non-class type 'const int'
         vec << src_pt.x, src_pt.y, src_pt.z, 1;
                                 ^
/share/CalibAnything/src/calibration.cpp:604:43: error: request for member 'z' in 'src_pt', which is of non-class type 'const int'
         vec << src_pt.x, src_pt.y, src_pt.z, 1;
                                           ^
/share/CalibAnything/src/calibration.cpp:608:30: error: request for member 'segment' in 'src_pt', which is of non-class type 'const int'
             int seg = src_pt.segment;
                              ^
/share/CalibAnything/src/calibration.cpp:620:32: warning: range-based 'for' loops only available with -std=c++11 or -std=gnu++11
         for (cv::Point point : lidar_points[i])
                                ^
CMakeFiles/LidarToCamera.dir/build.make:75: recipe for target 'CMakeFiles/LidarToCamera.dir/src/calibration.cpp.o' failed
make[2]: *** [CMakeFiles/LidarToCamera.dir/src/calibration.cpp.o] Error 1
CMakeFiles/Makefile2:110: recipe for target 'CMakeFiles/LidarToCamera.dir/all' failed
make[1]: *** [CMakeFiles/LidarToCamera.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

As I am no C++ programmer and have no particular knowledge of CMake I don't quite know where to start so any help will be appreciated.  

P.S: Is there any way to obtain pre-built binaries to use?

SST-bplus, 2024-06-12:
Ok so I was able to find the fault. The readme says: 

git clone https://github.com/OpenCalib/CalibAnything.git
cd CalibAnything
# mkdir build
mkdir -p build && cd build
# build
cmake .. && make

This clones from the repo https://github.com/OpenCalib/CalibAnything.git

However, for it to build in the provided container, the source from this repo

 https://github.com/PJLab-ADG/SensorsCalibration/tree/master/lidar2camera/auto_calib_v2.0

has to be built. So by skipping the first two lines of the instructions, the build succeeds.

Do you mind correcting the instructions for future users?

Thx!

zhangyuran-gg, 2024-10-16:
> 好的，所以我能够找到故障。自述文件说：
> 
> git clone https://github.com/OpenCalib/CalibAnything.git cd CalibAnything mkdir -p build & cd build cmake ..&& 制造
> 
> 此 clone 来自 repo https://github.com/OpenCalib/CalibAnything.git
> 
> 但是，要使其在提供的容器中构建，此存储库中的源代码
> 
> https://github.com/PJLab-ADG/SensorsCalibration/tree/master/lidar2camera/auto_calib_v2.0
> 
> 必须建造。因此，通过跳过说明的前两行，构建成功。
> 
> 您介意为未来的用户更正说明吗？
> 
> 感谢！

Hello, have you tested using your own dataset and what is the accuracy