PJLab-ADG/SensorsCalibration #163: lidar2car

ZXJohn, 2024-07-17:
when i run the example，it is occuring the problem：“Segmentation fault (core dumped)”。i found it is because the lidar_pose is：
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
Then lidar_pose_x_=0,lidar_pose_y_=0,Bspin_builder will fail.
Does anyone know how to solve it。

pedestrain123, 2024-11-28:
@ZXJohn 您好请问编译遇到这个错误怎么解决？/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bspline.cpp.o): relocation R_X86_64_32S against symbol `_ZTVN8SPLINTER10SerializerE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bsplinebasis.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bsplinebasis1d.cpp.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bsplinebuilder.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(datatable.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(function.cpp.o): relocation R_X86_64_32S against symbol `_ZTVN8SPLINTER8SaveableE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(mykroneckerproduct.cpp.o): relocation R_X86_64_32 against symbol `__gxx_personality_v0@@CXXABI_1.3' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(serializer.cpp.o): relocation R_X86_64_32S against symbol `_ZTVN8SPLINTER10SerializerE' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(utilities.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(datapoint.cpp.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIC
/usr/bin/ld: 最后的链结失败: 输出不可表示的节
collect2: error: ld returned 1 exit status

WangLu180, 2025-03-07:
the same,bro, have u solve it?

LeungWaiHo, 2025-05-29:
> [@ZXJohn](https://github.com/ZXJohn) 您好请问编译遇到这个错误怎么解决？/usr/bin/ld: /home/wdx/code/perception/sensor_calibarate/SensorsCalibration/SensorX2car/lidar2car/lib/libsplinter-static-3-0.a(bspline.cpp.o): [...]

I meet the same problem, have you solve it?