PJLab-ADG/SensorsCalibration #164: pull docker image失败

striving1234, 2024-07-18:
sudo docker pull scllovewkf/opencalib:v1
v1: Pulling from scllovewkf/opencalib
aa11c7d3aa8c: Retrying in 1 second 
d9a7d8445351: Retrying in 1 second 
0eaf53dd1034: Retrying in 1 second 
4d01e55a3979: Waiting 
2842279b3c74: Waiting 
error pulling image configuration: download failed after attempts=6: dial tcp [2a03:2880:f130:83:face:b00c:0:25de]:443: i/o timeout
在使用docker拉取image时超时。

striving1234, 2024-07-18:
sudo docker pull scllovewkf/opencalib:v1
Error response from daemon: Head "https://registry-1.docker.io/v2/scllovewkf/opencalib/manifests/v1": net/http: TLS handshake timeout

zlm646, 2024-08-01:
我也遇到这个问题，试了下把这些源都加上就ok了
"https://hub.uuuadc.top",
"https://docker.anyhub.us.kg",
"https://dockerhub.jobcher.com",
"https://dockerhub.icu",
"https://docker.ckyl.me",
"https://docker.awsl9527.cn",
"https://dockerproxy.com",
"https://docker.mirrors.ustc.edu.cn",
"https://docker.nju.edu.cn"

shaoxh, 2025-02-02:
# 创建目录

sudo mkdir -p /etc/docker

# 写入配置文件

sudo tee /etc/docker/daemon.json <<-'EOF'

{

    "registry-mirrors": [

    	"https://docker-0.unsee.tech",

        "https://docker-cf.registry.cyou",

        "https://docker.1panel.live"

    ]

}

EOF

# 重启docker服务

sudo systemctl daemon-reload && sudo systemctl restart docker